mantidproject/mantid #25881: Validation and system tests for ReflectometryISISLoadAndProcess

gemmaguest, 2019-06-12:
`ReflectometryISISLoadAndProcess` is a new algorithm which we aim to be the main workflow algorithm for reflectometry. We should validate it against existing system test results and write new system tests for it.

The main system tests we have at the moment are `INTERReductionTest.py` and `ISISReflectometryAutoreductionTest.py`. Both of these call `ReflectometryReductionOneAuto`, which is wrapped by `ReflectometryISISLoadAndProcess`, so initially we could create new tests which do a direct swap as far as possible with the new algorithm to check that it still produces the same results. It would be good then to get the new tests to exercise the additional functionality of `ReflectometryISISLoadAndProcess`:
- Do the loading within the algorithm.
- Pass transmission run numbers directly into the algorithm rather than calling `CreateTransmissionWorkspaceAuto` and `Stitch1D` separately.
- Pass slicing parameters into the algorithm rather than doing the slicing first manually.

There may be some differences in output workspace names which I think is ok, as long as the data is correct.

alicerussell1, 2019-07-26:
I started this on the branch `25881_validate_and_add_system_tests_for_refl`. The tests pass for the ones where I have replaced RROA with `ReflectometryISISLoadAndProcess` but the tests for slicing and transmission runs are failing.

gemmaguest, 2019-07-26:
> I started this on the branch `25881_validate_and_add_system_tests_for_refl`. The tests pass for the ones where I have replaced RROA with `ReflectometryISISLoadAndProcess` but the tests for slicing and transmission runs are failing.

@alicerussell1 	Ok, thanks for getting those first ones working. I'll create a draft PR for that and take it from there.

closed 2020-01-20 by gvardany