mantidproject/mantid #25899: Remove compatibility mode

TTitcombe, 2019-06-13:
ISIS SANS has a new reduction workflow, where events are kept for as long as possible. Because there is a slight discrepancy between data reduced the old and new ways, the gui has a compatibility mode to switch between workflows.

PR #25807 fixed a bug introduced when bin masking in the new workflow, minimising the difference between the workflows. 

The SANS scientists should evaluate the new workflow, with the aim of removing non-compatibility mode within a release or two if they find new the workflow acceptable.

rbauststfc, 2024-08-20:
During some recent investigations we discovered that this issue still hasn't been addressed. Currently compatibility mode is being applied by default to SANS reductions because it's not possible to control it via the TOML user file and the checkbox that controls it via the GUI is selected by default. Our scientists will do some testing to determine the following:

Short term:

- Should compatibility mode be switched off by default in the GUI?

Long term:

- Is there a use case for having compatibility mode any longer, as it was always intended to be temporary? If not then it should be removed entirely. If it is useful to keep then we should make it possible to set via the user file.

Note that when compatibility mode is switched off **and** the user is event slicing **and** they have ticked the `Optimize event slices` checkbox in the GUI, then the reduction diverts into a different reduction pathway (algorithm `SANSSingleReduction2` is used instead of `SANSSingleReduction`). This alternative reduction pathway changes the order of reduction steps slightly so that more are performed before any event slicing to try and speed up performance. Currently this reduction pathway is broken and the long-term decision on what to do about this will be looked at as part of wider discussions currently taking place on SANS performance issues. What this means, though, is that in the short term if we switch compatibility mode off by default then the 
`Optimize event slices` checkbox will go from having no effect to sending the user down a broken reduction route, so this checkbox should be hidden from the GUI while long-term decisions are made over the `SANSSingleReduction2` route.

rbauststfc, 2024-09-19:
After testing, our scientists have concluded that this feature should now be deprecated. For the next release we should untick the checkbox by default, and then long-term we should remove the feature altogether.