mantidproject/mantid #25908: Hard crash on help navigation

TTitcombe, 2019-06-14:
There is a hard crash in workbench's help window when you navigate to a page while you're looking at an image.

### Steps to reproduce the behaviour
In workbench, press F1 to bring up the mantid help page. Search for a page with an image e.g. `ISIS SANS`, and click on the link to bring up the page. Click on an image in the page to maximize it. Then click on any link in the search area to the left of the help window. Workbench hard-crashes

### Platforms affected
Ubuntu 18.04

hsautessella, 2019-06-21:
This isn't an issue on Windows as you can't open images in the same way you can in Ubuntu. I'll look into it

hsautessella, 2019-07-02:
This is also happening in MantidPlot

closed 2021-03-02 by stale[bot]