mantidproject/mantid #25915: Fix broken SANSILLReduction unit test

peterfpeterson, 2019-06-14:
4e4ffc7e331d610b593d84f92b3206f275606a5b disabled a unit test that was [failing on all build servers](https://builds.mantidproject.org/job/master_incremental/4579/). Re-enable the test, figure out why it is failing, and fix it.

gvardany, 2019-06-18:
Git bisect points to fc57517a9643c337c2a6a01d988ce3923ea0f012 as the first bad commit.
Indeed, 84793429bf86cbc7736b3bb007e519b53b3078db is still fine.
@peterfpeterson Any idea how that innocent change can cause the output of `CalculateTransmission` not have instrument attached?

gvardany, 2019-06-18:
It turns out that after implementing `__len__`, the instrument, which is a component assembly, gets boolean evaluation as a container, so empty container leads to False:
From [what is False](https://thomas-cokelaer.info/tutorials/python/boolean.html):
```
Objects from classes that implements __len__ to return False or zero
```
Any idea how to resolve this?

peterfpeterson, 2019-06-18:
We could implement `__bool__` explicitly for instrument to return what one would expect

gvardany, 2019-06-18:
Yes we should. I fixed the broken test however to use `isinstance`.

PR #25924 (2019-06-18): Fix broken test 
**Description of work.**

This PR fixes and re-enables the broken unit test for ILL SANS.
Also reverts 2cbd096 and 63b6a31 from PR #25307, that were poorly done. 
Issue [#25923](https://github.com/mantidproject/mantid/issues/25923) will address it properly.

**To test:**
`ctest -R SANSILLReduction` should pass.

Fixes #25915 . 

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/IndividualTicketTesting/)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the code conform to the [coding standar [... cut]
closed 2019-06-18 by martyngigg