mantidproject/mantid #25916: Set date for Old ISIS SANS removal

TTitcombe, 2019-06-17:
The old ISIS SANS gui was deprecated in Mantid 4.0.
A date/release number for removing the old gui should be agreed by the scientists.

A message should be posted to the logger when you load the Old gui, which clearly states the last version it will be present in, and that users should migrate to `ISIS SANS` (the new gui)

gemmaguest, 2019-08-29:
The deprecation notice was added in 4.1 (although no date for removal was specified) so we can aim to remove it in 4.2 if scientists are happy. There may be a possible issue with backward compatibility of User Files - waiting on SANS Team.

smk78, 2020-07-16:
On review, the SANS Group feels it needs some understanding of what the consequences of removing the Old Backend might be; for example, @gemmaguest mentions possible backward compatibility issues with User Files?