mantidproject/mantid #25926: Mantid Setup Window - Do not show again box doesn't work

robertapplin, 2019-06-18:
### Description
The checkbox at the bottom left of the Mantid setup screen called `Do not show again until next release` does not work. Even when ticked, the setup screen always appears whenever opening mantid.

### Platforms affected
Windows and maybe others

DanNixon, 2019-07-01:
You have to have a facility and instrument set. I noticed this a few times before I realised that was required.

I'm not sure this is a sensible requirement though? Facility is likely needed for archive search to function correctly, but instrument is probably not important.

robertapplin, 2019-07-01:
I agree I am not sure about this requirement. Maybe the option should be invisible if the instrument or facility is not set?

DanNixon, 2019-07-01:
If Mantid is deemed to be not configured if no facility is selected then we shouldn't allow a user to close this window without selecting one.

I think not selecting an instrument should be allowed. AFAIK most of Mantid either doesn't care, can determine from data or will ask more locally about the instrument.

PR #26126 (2019-07-02): First Time Setup - Do not show until next release remove instrument check
**Description of work.**
This PR ensures that the `Do not show until next release` checkbox in the First Time Setup window will stop the window from opening until the next release, when loading mantid.
It was decided that the check on the instrument was not important as certain parts of mantid can still be used without setting an instrument. 
Previously, checking the `Do not show until next release` checkbox would do nothing if an instrument was not selected. This caused confusion with some users, and looks as if the opt [... cut]
closed 2019-07-02 by martyngigg