mantidproject/mantid #25946: Muon Analysis 2: Setting instrument to None breaks other interface

martyngigg, 2019-06-19:
### Expected behavior

Changing instrument to None simply resets things.

### Actual behavior

Changing the instrument to None, closing the Muon GUI and reopening it gives the following traceback:

```
Traceback (most recent call last):
  File "/opt/mantidnightly/lib/workbench/app/mainwindow.py", line 370, in <lambda>
    action.triggered.connect(lambda checked_py, script=script: self.launch_custom_python_gui(script))
  File "/opt/mantidnightly/lib/workbench/app/mainwindow.py", line 347, in launch_custom_python_gui
    self.interface_executor.execute(open(filename).read(), filename)
  File "/opt/mantidnightly/lib/mantidqt/widgets/codeeditor/execution.py", line 172, in execute
    exec (code_obj, self.globals_ns, self.globals_ns)
  File "/opt/mantidnightly/scripts/Muon_Analysis_2.py", line 23, in <module>
    muon_analysis = MuonAnalysisGui()
  File "/opt/mantidnightly/scripts/Muon/GUI/MuonAnalysis/muon_analysis_2.py", line 67, in __init__
    self.data_context = MuonDataContext('Muon Data', self.loaded_data)
  File "/opt/mantidnightly/scripts/Muon/GUI/Common/contexts/muon_data_context.py", line 80, in __init__
    self._instrument = ConfigService.getInstrument().name() if ConfigService.getInstrument().name()\
RuntimeError: Failed to find an instrument with this name in any facility: '' - search object
```

Note that this also breaks the settings GUI under `File->Settings` with the same error.

### Steps to reproduce the behavior

On master:
* start workbench
* open Muon Analysis 2
* select instrument `None`
* close Muon Analysis 2
* reopen Muon Analysis 2
* :boom: 

### Platforms affected

All

### Workaround

The current workaround is to run

```python
from mantid import config
config['default.instrument'] = ''
```
in the editor or IPython console.

martyngigg, 2019-06-19:
Question: Does the `None` option need to exist at all?

AnthonyLim23, 2019-06-20:
I thought we got rid of it.

martyngigg, 2019-06-20:
It's still there in nightly.

Mantid-Matthew, 2019-07-05:
The option has now been removed from the GUI

closed 2019-07-05 by Mantid-Matthew