mantidproject/mantid #26020: py3compat Enums have different behaviour on OSX

TTitcombe, 2019-06-27:
### Expected behavior
The `mantid.py3compat.Enum` class should 1) mirror python3's Enum 2) behave the same on every OS

### Actual behavior
PR #26017 had failing tests on the OSX builds (https://builds.mantidproject.org/job/pull_requests-osx/30335/testReport/junit/projectroot.scripts.test.SANS/gui_logic/PythonSANSQt4_gui_state_director_test_gui_state_director_test/).
It appears that Mantid's python2 implementation of Enums behave differently on OSX.

On ubuntu and windows, an instance of an enum is a class `<enum 'EnumName'>`, so `isinstance(EnumName.someInstance, EnumName)` returns True.

On OSX, `EnumName.someInstance` is an `EnumValue` class. `isinstance(EnumName.someInstance, EnumName)` returns False.

It is not clear yet whether this discrepancy in behaviour is present in python3 Enums, or whether our python2 implementation is incorrect

### Steps to reproduce the behavior
```
from mantid.py3compat.enum import Enum, EnumValue
MyEnum = Enum("MyEnum", "foo bar")
foo = MyEnum.foo
print(isinstance(foo, MyEnum))
type(foo)
type(MyEnum)
```
This should print 
```
True
<enum 'MyEnum'>
enum.EnumMeta
```
for both a python2 and python3 build of Mantid

### Platforms affected
OSX? Python2 and 3?

TTitcombe, 2019-06-27:
**Update:**
It also appears this affects RHEL 7: https://builds.mantidproject.org/job/pull_requests-rhel7/30291/testReport/junit/projectroot.scripts.test.SANS/state/python_SANS_state_save_test_save_test/

ISIS SANS serializes and deserializes Enums as part of the "SANS state" object which is passed between algorithms. The serialization turns the enum into a string containing the module from which the enum was imported, the name of the enum, and the enum value. Deserializing uses `importlib.import_module` and `getattr` to return the Enum.

It is possible that the issue is with the (de)serialization, not with enums

DavidFair, 2020-02-25:
Closing as fixed by #27541

closed 2020-02-25 by DavidFair