mantidproject/mantid #26025: Reflectometry GUI restructuring: fixes and improvements

gemmaguest, 2019-06-27:
Remaining work and suggestions for the restructured reflectometry GUI. This includes remaining tasks from #23051 and #23027.

### Important
- [ ] Thorough stability testing
- [ ] Documentation should be checked and updated where necessary, particularly new keyboard shortcuts
- [ ] Unit tests are missing for MainWindowPresenter
- [ ] Unit tests are missing for cutting and pasting rows in the runs table (Notably Clipboard.h)
- [ ] Unit tests are missing for default Instrument and Experiment options when not in the instrument parameters file
- [ ] Cross check tests against old GUI tests which were removed

### Should have
**Table**
- [ ] Add a group and row when the GUI starts
- [ ] Add an empty row to the empty group when the table is cleared
- [ ] Remove the empty group when starting autoprocessing

**Search**
- [x] #25970 Fix catalog login for workbench (CatalogLoginDialog fails)
- [ ] Autoprocessing should not start if catalog login fails or if the search string is empty.

**General**
- [x] Row highlighting is lost during processing
- [ ] Disabled icons should be greyed out
- [ ] If you create a new batch, close batch 0, and then create a new batch, you'll have 2 batch 1 tabs.
- [ ] Text in Event Handling, Experiment Settings, and Instrument Settings tabs are a lot smaller than Runs and Save ASCII (Ubuntu 18)
- [ ] In Event Handling -> Uniform slicing you can type a + at the start of the number. For even time slices you cannot type a - at all, but for (sec) slices you can type a - but no numbers after it. This doesn't cause any errors but is confusing and annoying so it would be good to sort out if possible.
- [ ] As noted on PR #25798 it would be good to move `ModelCreationHelper` to a `TestHelpers` subdirectory.

### Would be nice to do
- [ ] #23067 - Makes Copy+Paste applicable in more circumstances, adds ability to paste rows from one group into another group. *Does not require modification to `JobTreeView` copy+paste*.
- [ ] #23032 - Enables nicer automatic save
- [ ] #23049 - Better handling of erroneous input.
- [ ] #23036 - More reliable theta value loading, Improved runs table usability.
- [ ] #23039 - Improved runs table usability.
- [ ] Globally define colours e.g. for invalid inputs, rather than hard-coding them in each view.
- [ ] Check if it's worth tidying up duplicated code such as the `toCsv` and `messageFor` functions which appear in both ExperimentView.cpp and InstrumentView.cpp.
- [ ] Check if it's worth refactoring repeated code like `acceptMainPresenter` into a base class for the tab presenters, and whether we could have a single interface e.g. `ITabPresenter` instead of separate interfaces for each tab presenter.

NickDraper, 2019-07-10:
Is everything for R4.1 in this issue done now?  Should the remains of this umbrella issue move to R4.2?

gemmaguest, 2019-07-10:
@NickDraper It was just documentation left so I've created #26282 for that and moved this to 4.2

closed 2021-03-02 by stale[bot]