mantidproject/mantid #26078: Unscripted Testing ISIS Indirect interfaces

NickDraper, 2019-07-02:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

Follow the directions at: http://developer.mantidproject.org/Testing/IndirectInelastic/IndirectInelasticAcceptanceTests.html

LolloB, 2019-07-03:
Documentation page for indirect testing does not correctly display images, see #26142.

LolloB, 2019-07-03:
ISIS Calibration in Data reduction interface allows plotting raw with invalid data, see #26149

LolloB, 2019-07-03:
When fitting ElasticDiffRotDiscreteCircle MantidPlot crashes, see #26154

LolloB, 2019-07-03:
When using the same file as input and corrections in Indirect Corrections, MantidPlot crashes, see #26157

LolloB, 2019-07-03:
When one tries to close the Indirect setting interface, MantidPlot asks to save current changes and then closes, see #26158

LolloB, 2019-07-03:
When trying to plot the results of Indirect Simulation MolDyn MantidPlot crashes, see  #26164

LolloB, 2019-07-04:
Issues #26157 and #26164 affect the workbench as well

LolloB, 2019-07-04:
Indirect Bayes interface invalidates resolution workspace on MantidPlot, see #26173

LolloB, 2019-07-04:
Indirect Symmetrise fails when EMax=0.4, see #26179

LolloB, 2019-07-04:
Background range can be set in Data Analysis Elwin even if background is disabled, see #26182

LolloB, 2019-07-04:
Calculate Paalman Pings crashes the workbench, see #26183

LolloB, 2019-07-04:
Indirect diffraction initializes to the wrong number of spectra, see #26188

AnthonyLim23, 2019-07-18:
looks like I will probably do most of this tomorrow.

AnthonyLim23, 2019-07-18:
#26385 is a crash in data reduction that makes mantid unusable

LolloB, 2019-07-18:
I went through the tests on Ubuntu 18.04 and found the following:
- Peak selector lines in Indirect Bayes move only in one direction #26393
- Closing Indirect interfaces causes mantid to crash #26396

closed 2019-07-19 by NickDraper