mantidproject/mantid #26081: Unscripted Testing Documentation

NickDraper, 2019-07-02:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

In the build directory run `cmake docs-html` 
Check in the `docs/html` directoy:

* Algorithm, fit, concept and api pages should be generated
* Algorithm dialog snapshots should appear on algorithm pages in offline help
* Math formulae should appear on algorithm pages in offline help
* workflow diagrams should appear on algorithm pages in offline help

Pasarus, 2019-07-09:
@LolloB did you manage to get this done?

LolloB, 2019-07-10:
Yes, I have tested it on Ubuntu 18.04 and everything seems to be correct.

Pasarus, 2019-07-19:
- app.verbose() is now deprecated. Use sphinx.util.logging instead. in Broadcast Workspace documentation
- ```Warning: Unable to reclaim box space in spline routing for edge "subtractCanLAB" -> "valQLAB". Something is probably seriously wrong.``` In SANSStitch-v1.rst
- algorithms/PDConvertRealSpace-v1:8: WARNING: Duplicate ID: "equation-gofr".
- Muon Elemental Analysis.rst:51: WARNING: Could not lex literal_block as "json". Highlighting skipped.
- /docs/source/release/v3.11.0/framework.rst:53: WARNING: undefined label: algm-savemdworkspacetovtk-v1 (if the link has no caption the label must precede a section header)
- /docs/source/release/v3.11.0/framework.rst:53: WARNING: undefined label: algm-loadvtk-v1 (if the link has no caption the label must precede a section header)
- /docs/source/release/v3.7.1/framework.rst:141: WARNING: undefined label: algm-savemdworkspacetovtk (if the link has no caption the label must precede a section header)
- /docs/source/release/v3.9.0/framework.rst:35: WARNING: undefined label: algm-savemdworkspacetovtk (if the link has no caption the label must precede a section header)
- /docs/source/release/v3.9.0/ui.rst:60: WARNING: undefined label: algm-savemdworkspacetovtk (if the link has no caption the label must precede a section header)

AndreiSavici, 2019-07-21:
@Pasarus The warnings about `algm-savemdworkspacetovtk` means that you've built Mantid without paraview. In that case, `LoadVTK` and `SaveMDWorkspaceToVTK` are not built, therefore no documentation for those. So you cannot link to that.

closed 2019-07-22 by Pasarus