mantidproject/mantid #26084: Unscripted Testing Mantidplot Group 4

NickDraper, 2019-07-02:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

See https://www.mantidproject.org/Unscripted_Manual_Testing#Groups

hsautessella, 2019-07-03:
#### Tested on Windows 10 running nightly v4.0.21090702.2018
### Qt Help Documentation
- Opening an image and then selecting a page within the search bar crashes the help window: #25908

### Catalog (icat) searching & downloading
- The help button on catalog search directs to online help rather than offline (not sure if this is worth raising an issue for)
- Catalog throws unhandled exception when opening 0 byte file #24998

NickDraper, 2019-07-19:
Time has run out to attempt this testing

closed 2019-07-19 by NickDraper