mantidproject/mantid #26085: Unscripted Testing Mantidplot Group 3

NickDraper, 2019-07-02:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

See https://www.mantidproject.org/Unscripted_Manual_Testing#Groups

ciaranightingale, 2019-07-02:
Testing against Windows 10 64 bit

ciaranightingale, 2019-07-02:
Pressing the question mark on the about page causes a red stop symbol to appear which does not disappear until you click something else (the question mark button therefore doesn't do anything)

NickDraper, 2019-07-02:
Can you create an issue for this, 
Set the milestone as "Release 4.2"
Add an Induction 1 label, and while you are at it assign it to yourself

Then you can close this issue

ciaranightingale, 2019-07-03:
Issue created.

closed 2019-07-19 by ciaranightingale

ciaranightingale, 2019-07-19:
Windows 10, no issues to report.