mantidproject/mantid #26088: Unscripted Testing Muon

NickDraper, 2019-07-02:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

See instructions at http://docs.mantidproject.org/v3.9.0/interfaces/Muon_Analysis_unscripted_testing_guide.html
http://developer.mantidproject.org/Testing/MuonInterface/MuonTesting.html

Harrietbrown, 2019-07-02:
Testing in Wondows 7/8/10

Harrietbrown, 2019-07-02:
https://docs.mantidproject.org/v4.0.0/interfaces/Muon%20Analysis%20Unscripted%20Testing%20Guide.html
Is the tests for the current version

closed 2019-07-18 by Harrietbrown