mantidproject/mantid #26089: Unscripted Testing TOFConverter

NickDraper, 2019-07-02:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

Make sure that inputs and outputs work sensibly, stress test with some bad inputs (e.g. letters in a numeric input)
See http://docs.mantidproject.org/v3.12.0/interfaces/TOF%20Converter.html

NickDraper, 2019-07-02:
Lorenzo, could you test this please.  It needs checking on Mantidplot and the workbench, they look similar but have completely different code.

LolloB, 2019-07-02:
In MantidPlot the TOF converter accepts some non-numeric values ('e', '.' and '-') alone only returning a traceback instead of opening a dialog window to enforce positive floats.
(Tried on Ubuntu 18)

LolloB, 2019-07-02:
Help button in mantidplot TOF converter returns a broken lins:
Missing page - qthelp://org.mantidproject/doc/interfaces/TOF Converter.html

LolloB, 2019-07-18:
Testing Mantid 4.0.20190718.1003 on Ubuntu 18.04.

On workbench:
- inputs containing the letter 'e' are accepted even when not used as `<val>e<exp>`. This will raise an exception insead of opening a warning box:
>Traceback (most recent call last):
  File "/home/lorenzo/mantid/scripts/TofConverter/converterGUI.py", line 118, in convert
    if float(self.ui.InputVal.text()) <= 0:
ValueError: could not convert string to float: e12

- Giving fractional exponents (eg. `1e0.9`) raises an exception instead of opening a warning box:
> Traceback (most recent call last):
  File "/home/lorenzo/mantid/scripts/TofConverter/converterGUI.py", line 118, in convert
    if float(self.ui.InputVal.text()) <= 0:
ValueError: invalid literal for float(): 1e0.9

On MantidPlot:
- inputs containing the letter 'e' are accepted even when not used as `<val>e<exp>`. This will raise the same exception as the Workbench
- Giving fractional exponents (eg. `1e0.9`) raises the same exception as the workbench.

Apart from this, everything else seems to be working as intended.

closed 2019-07-18 by LolloB