mantidproject/mantid #26098: Unscripted Testing Engineering Diffraction

NickDraper, 2019-07-02:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

Follow: http://docs.mantidproject.org/nightly/interfaces/Engineering%20Diffraction%20Test%20Guide.html

LolloB, 2019-07-02:
Opened issue #26115, RB Numbers are not validated correctly

LolloB, 2019-07-02:
When trying to focus data and providing it a random file it will fail with the following error:
Error in execution of algorithm EnggFocus
Unable to find a meaningful list of workspace indices for the bank passed: 2. Please check the inputs.
  at line 125 in '/home/lorenzo/mantid/Framework/PythonInterface/plugins/algorithms/EnggFocus.py'
  caused by line 105 in '/home/lorenzo/mantid/scripts/Engineering/EnggUtils.py'
Produced focused workspace: engggui_focusing_output_ws_bank_2
The focusing failed. Some input properties were not valid. See log messages for details. Error: Invalid value for property InputWorkspace (MatrixWorkspace) from string "engggui_focusing_output_ws_bank_2": Workspace "engggui_focusing_output_ws_bank_2" was not found in the Analysis Data Service
The focusing did not finish correctly. Check previous log messages for details
ValueError: Workspace 'engggui_focusing_output_ws_bank_2' not found!
  at line 3 in '<Interface>'
  caused by line 257 in '/home/lorenzo/mantid-build/bin/pymantidplot/__init__.py'
  caused by line 794 in '/home/lorenzo/mantid-build/bin/pymantidplot/proxies.py'

Is this expected or should it pop up a window to the user?

LolloB, 2019-07-02:
Help button returns broken link:
  _qthelp://org.mantidproject/doc/interfaces/Engineering Diffraction.html_

LolloB, 2019-07-02:
When loading example data the fitting function fails:
_Could not fit any peak.  Failed to fit peaks with peak type BackToBackExponential even though FindPeaks found 2 peaks in principle. See the logs for further details._

LolloB, 2019-07-02:
As I could not find any peaks, I could not test the GSAS-II Refinement.
In the previous parts I could not find other issues.

LolloB, 2019-07-18:
The following issues were found in Ubutnu 18.04:
- Engineering diffraction stop focus crashes #26383
-  Pre-processing bad data in Engineering diffraction causes a hard crash #26384 
Like before the fitting function failed so I was unable to test the `GSAS-II Refinement` tab.

closed 2019-07-18 by LolloB