mantidproject/mantid #26100: Unscripted Testing Workspace Groups

NickDraper, 2019-07-02:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

* Follow the examples at http://docs.mantidproject.org/nightly/concepts/WorkspaceGroup.html 
* Ensure MantidPlot displays Group Workspaces correctly in the workspace tree

alicerussell1, 2019-07-03:
Tested on windows. Bugs found:
- If you group workspaces inside another group, it creates a new separate group with a copy of the workspaces inside. This means there are two workspaces with the same name which shouldn't be allowed. #26151
- Grouping a workspace group with the name NewGroup will cause an error due as it is trying to create a workspace group with the same name #26137 
- The script in nested workspace group in https://docs.mantidproject.org/nightly/concepts/WorkspaceGroup.html does not work in the script window. #26152

alicerussell1, 2019-07-18:
Tested on windows. No new bugs found

closed 2019-07-18 by alicerussell1