mantidproject/mantid #26102: Unscripted Testing Sample Transmission calculator

NickDraper, 2019-07-02:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

Make sure that inputs and outputs work sensibly, stress test with some bad inputs (e.g. letters in a numeric input)
See http://docs.mantidproject.org/v3.7.2/interfaces/SampleTransmissionCalculator.html

hsautessella, 2019-07-04:
#### Tested on Windows 10 Nightly v4.0.20190702.2018

- Hard crash when setting plot options after a failed plot #26172

hsautessella, 2019-07-18:
#### Tested on Windows 10 Nightly 4.0.210390717.2323
- Negative values allowed for wavelength in multiple input mode: #24224
- Hard crash when setting plot options after a failed plot #26172
- Output plot's y-axis ticks sometimes disappear when changing to log scale  #24981

closed 2019-07-18 by hsautessella