mantidproject/mantid #26103: Unscripted Testing ISIS Reflectometry (old and new)

NickDraper, 2019-07-02:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

For testing procedures see http://www.mantidproject.org/Testing_ISIS_Reflectometry_GUI

DTasev, 2019-07-02:
The documentation is somewhat out of date from the ISIS Reflectometry new², so I was making some stuff up as I went, but I tried to follow the ideas

#### Crash when plotting after processing - Workbench ONLY

---------

@Pasarus is fixing/has fixed this by the time this comment is visible

---------

- Open GUI, switch to `POLREF`, put in investigation ID 1520249
- Select run 14885 and 14886, Transfer them in
- Click the group and process
- Click the "Plot selected rows" button while having the group selected, Workbench should crash
```
********* UNHANDLED EXCEPTION *********
  what(): Unexpected type: 'WorkspaceGroup'
  at line 60 in '/home/dtasev/dev/m/qt/python/mantidqt/plotting/functions.py'
  caused by line 203 in '/home/dtasev/dev/m/qt/python/mantidqt/plotting/functions.py'
```

#### Cut crashes MantidPlot & Workbench
- Open GUI, switch to `POLREF`, put in investigation ID 1520249
- Select runs 14885-14893, Transfer them in
- Select the bottom **group** one and Cut
- Try to transfer 14893 again

2nd way of doing this

- Open GUI, switch to `POLREF`, put in investigation ID 1520249
- Select runs 14885-14893, Transfer them in
- Select any of the middle **groups** and Cut

#### Unhandled exception with invalid Python slicing list
- Open GUI, switch to `POLREF`, load run 14885, angle 0.5
- Go to `Event Handling`, put `biscuits` in the `Python list (sec)` field
- Go back to `Runs`, click process, unhandled exception "Invalid slicing" will occur
  - As a bonus: if you `Continue` and try to close MantidPlot it will give you a "Window cannot be closed" error, before closing
  - Workbench will hard crash
- This also applies to the `Python list` below
  - Putting in "biscuits" will give the unhandled exception "Invalid slicing"
  - Putting in "1,2,3" will give another unhandled exception "Custom log values are not implemented..."
  - Note that both of these are unhandled and so will hard-crash the Workbench

#### Save ASCII tab - can't see `Title` and `Q resolution` changing the output files
- Open GUI, switch to `POLREF`, load run 14885, angle 0.5
- Process it
- Go to Save ASCII tab
- Save out different formats with `Title` and `Q resolution` being on/off

#### Save ASCII tab - `.mft`, and the two `.dat`s seem to insert the separator on the start of every line
- Open GUI, switch to `POLREF`, load run 14885, angle 0.5
- Process it
- Go to Save ASCII tab
- Save out different formats with different separators

I am not sure whether this breaks anything, but usually the separator is not added at the start of the lines.

- The ANSTO `.txt` one doesn't have a separator on the start of each line

gemmaguest, 2019-07-02:
- The crashes to do with Cut are fixed on master
- The Invalid Slicing error is fixed on master but the "Custom log values not implemented" one is not - I've raised #26127 for this
- The issues to do with saving are existing known problems and will be looked at when we do #24429

DTasev, 2019-07-03:
Old Reflectometry GUI
---------------------

#### Options doesn't seem to keep the option for "Accumulation method" regardless of whether `Cancel` or `OK` are pressed

#### Status bar doesn't show "Processing row X"
- It just sits on `Ready` all the time

#### Pressing the `X` button shows the exit dialog in the top left
- Pressing Ctrl+F4 or Close Refl GUI shows it in the middle

#### Save as 3 Column prints out "Not yet implemented"

--------

Didn't find any crashes, the ones above are just some bug fixes that probably aren't worth doing, as most of those things are likely fixed in the new² GUI

DTasev, 2019-07-03:
@gemmaguest When you've had a chance to read the old gui bugs and decided what/if action should be taken, could you please close this ?

gemmaguest, 2019-07-08:
The issues with the old GUI are all existing, minor problems, so are not relevant to this release

DTasev, 2019-07-18:
NEW GUI
----------

Minor issues, not for 4.1:

#### Pause processing of runs is enabled after opening the GUI, clicking it once makes it disabled

#### Processing a group with multiple rows never makes the progress bar 100%

#### Some deprecation warnings are being shown when savings things from the Save ASCII tab
```
SaveReflCustomAscii is deprecated (on 2018-06-29). Use SaveReflectometryAscii instead.
SaveReflCustomAscii started
SaveReflCustomAscii successful, Duration 0.01 seconds
SaveILLCosmosAscii is deprecated (on 2018-06-29). Use SaveReflectometryAscii instead.
SaveILLCosmosAscii started
SaveILLCosmosAscii successful, Duration 0.00 seconds
SaveANSTOAscii is deprecated (on 2018-06-29). Use SaveReflectometryAscii instead.
SaveANSTOAscii started
SaveANSTOAscii successful, Duration 0.01 seconds
```

#### Saving a file with a unicode character in it pre-pends `���` and appends `(invalid encoding)`
- Set prefix to óóó
- Save workspace
- Workspace is saved but filename looks something like this: `���TOF_14885_1.dat (invalid encoding)`

@Pasarus

Pasarus, 2019-07-18:
@DTasev I am attempting to reproduce Processing a group with multiple rows never makes the progress bar 100% However it works for me.... 
[image]

Pasarus, 2019-07-19:
@DTasev I also had it save to ascii file successfully 
[image]

closed 2019-07-19 by DTasev