mantidproject/mantid #26104: Unscripted Testing ISIS SANS (v2 GUI)

NickDraper, 2019-07-02:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

# SANS GUI Testing

<div class="contents" data-local="">

</div>

## Data reduction

*Preparation*

  - Get the ISIS sample data from the website
  - Ensure that the ISIS sample data directory is in the search
    directories

**Time required 20 - 30 minutes**

-----

1.  Open `Interfaces` \> `SANS` \> `SANS v2`

2.  Set `Instrument` to `LOQ`

3.  Choose `Load User File`; from the ISIS sample data in the `loqdemo`
    folder, choose `Maskfile.txt`

4.  Choose `Load Batch File`; from the ISIS sample data in the
    <span class="title-ref">loqdemo</span> folder, choose
    `batch_mode_reduction.csv`

5.    - In the `Settings` tab:
        
          - Go to `Mask`
          - Click `Display Mask`
          - This should give an instrument view with a circle at the
            centre

6.    - In the `Runs` tab
        
          - Click `Process`
          - After some seconds the rows should turn green
          - In the Main window there should be a series of new
            workspaces; 3 group workspaces and 4 2D workspaces
          - Change the first column of the first row to 74045; click
            process
          - The row should turn blue; hovering over the row should give
            an error message
          - Change the `Reduction` button to 2D
          - Check the `Plot Results` box
          - Click `Process`
          - A plot window will open; initially empty, then with a red
            line
          - Change `Reduction` back to 1D
          - Click `Process`
          - In the plot you should end up with a red and a black line
            plotted
          - Check the `Multi-period` box
          - 6 additional rows should appear in the table
          - Check that the `Add`, `Delete`, `Copy`, `Paste`, `Cut` and
            `Erase` icons work as expected
          - Delete all rows and re-load the batch file as previously

7.    - In the `Settings` tab
        
          - Set `Reduction Type` to `Merged`
          - Return to the `Runs` tab
          - Ensure `Plot results` is checked
          - Click `Process`
          - This should result in a plot with three lines

8.    - In the `Beam centre` tab
        
          - Click
          - A plot should appear after some seconds, with 4 lines
          - The 4 lines should gradually get closer together
          - This will run for some time, probably minutes

9.    - In the `Sum Runs` tab
        
          - Enter `74044, 74019` in the top line
          - Enter `LOQ74044-add` as Output file
          - Ensure that the `Output directory` is in you r managed paths
          - Click `Sum` at the bottom
          - Go back to the `Runs` tab
          - Remove all rows
          - Reload the batch file as before
          - Change the first column of both rows to `LOQ74044-add`
          - Click `Process`
          - This should now process as before

10.   - In the `Diagnostic Page` tab
        
          - For run choose `Browse` and load the `LOQ74044.nxs` file
          - Click each of the `Integral` buttons
          - They should produce plots
          - Check the `Apply Mask` boxes and click the buttons again
          - They should produce new, slightly different plots

11.   - In the `Runs` tab
        
          - Check all variables and inputs for tool tips by hovering
            over them

LolloB, 2019-07-04:
Testing on MantidPlot for Ubuntu 18 the following problems were found:
- Missing hints in ISIS SANS v2, see #26196

LolloB, 2019-07-05:
Tested on Workbench for Ubuntu 18, the following problems were found:
- Display Masks fails, see #26202
- Beam Centre does not display a plot, see #26203

LolloB, 2019-07-18:
Tested on MantidPlot and Workbench for Ubuntu 18.04.
The only issue found is #26196.
Everything else seems to work as intended.