mantidproject/mantid #26106: Unscripted Testing - Mantid Basics Course

NickDraper, 2019-07-02:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

* Build a docs-html target of Mantid
* Check against the online documentation (http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/index.html#mantid-basic-course)
* Open up the basic course (docs/html/tutorials/mantid_basic_course/index.html)
* Check that the pages in there make sense

robertapplin, 2019-07-02:
I have done half the testing so far, and a few issues have been spotted during this testing #26116 and #25926 . Tested on windows.

NickDraper, 2019-07-19:
Time has run out to attempt this testing

closed 2019-07-19 by NickDraper