mantidproject/mantid #26107: Unsrtipted testing M-slice

NickDraper, 2019-07-02:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

See https://mantidproject.github.io/mslice/ and follow Quick Start

alicerussell1, 2019-07-04:
Testing on windows 10. Bugs found:
- If you save a file in a `.txt` format using save button and try and load it in data loading, it returns an uncaught error. note: it does not happen if you save as a `.nxs` file. [#481](https://github.com/mantidproject/mslice/issues/481) 
- Save Cut to Workspace returns error if no cut is selected [#480](https://github.com/mantidproject/mslice/issues/480)
- On slice tab in workspace manager it returns a runtime error if you try and set the axis to |Q| and 2Theta [482](https://github.com/mantidproject/mslice/issues/#482)
-If you plot a workspace on the `MD histo` tab and in the settings tab enter an 'unknown symbol' in the x or y axis name i.e. $\mathrm{\\AA}$ (an extra \) then an error is thrown and the entire plot vanishes, leaving an empty window with one axis remaining [#483](https://github.com/mantidproject/mslice/issues/483)
-Error in script in documentation [#485](https://github.com/mantidproject/mslice/issues/485)

LolloB, 2019-07-18:
Testing on Ubuntu 18.04 I could find no issues.

closed 2019-07-19 by NickDraper