mantidproject/mantid #26120: Muon Analysis 1 resutlts table

Harrietbrown, 2019-07-02:
### Expected behavior
Results table in muon analysis 1 should output table of log values, converting the type into a string as needed

### Actual behavior
When creating table using sample_temp or run_number, mantidplot returns a type mismatch error

### Steps to reproduce the behavior
1. Open mantidplot
1. Open muon analysis
1. Load EMU 20918
1. Go to Data Analysis tab
1. Add a flat background and Abragam to fitting browser
1. Fit
1. In runs load EMU 20918-20
1. Without changing any settings use Fit again
1. Go to results table
1. Check run_number and sample_temp, and check both fitting results
1. Create Table

### Platforms affected
Tested on windows.

AnthonyLim23, 2019-07-03:
I think I understand why this happens. The second fit is on a co-added set of runs, the old GUI outputs the logs as a range (if they are not the same for all of the runs that have been co-added). Whereas the first fit was on a single run and has just a numerical value.

AnthonyLim23, 2019-07-03:
here we set the column types https://github.com/mantidproject/mantid/blob/master/qt/scientific_interfaces/Muon/MuonAnalysisResultTableCreator.cpp#L130

here we set the row values:
https://github.com/mantidproject/mantid/blob/master/qt/scientific_interfaces/Muon/MuonAnalysisResultTableCreator.cpp#L508

I think its the second one thats gone wrong as it only checks the current log value and not the column type. However, changing this could lead to problems in the first link if it is not checking every ws for a given log value.

PR #26200 (2019-07-08): Results  table fix for muon analysis 1
Fixes a bug in muon analysis results table that causes the table to fail if the column values are a mix of ranges and values.

**To test:**

Open mantidplot
Open muon analysis
Load EMU 20918
Go to Data Analysis tab
Add a flat background to fitting browser
Fit
Go to results table
Check run_number and sample_temp, and check both fitting results
Create Table
You should be able to plot sample_temp and the flat background value
Go back to the data analysis tab
In runs load EMU 20918-20
Without changing any settings use Fit again
Go to results table
Check run_n [... cut]
closed 2019-07-08 by NickDraper