mantidproject/mantid #26136: Instrument view miniplot save to workspace no longer works

martyngigg, 2019-07-03:
**Original reporter:** Pascal ISIS

### Expected behavior
You can save an integrated plot from the instrument view miniplot to a workspace

### Actual behavior
Clicking "Save to workspace" does nothing

### Steps to reproduce the behavior

Observed with WISH data. Load a file, show the instrument. 

Choose the "pick" tab, choose an ellipse, the integrated intensity appears and in the window where it appears, when we right click and select "Save Plot to workspace" that bit doesn't work for us.

### Platforms affected
All

ConorMFinn, 2019-07-03:
Also observed with GEM, HRPD, MARI. Works as expected on SXD and LOQ instruments.

PhilColebrooke, 2019-07-03:
I've also found that using the "Add single crystal peak" tool and doing "Save plot to workspace" causes a hard crash in Workbench.

martyngigg, 2019-07-04:
@ConorMFinn @PhilColebrooke Thank you for the additional information here. I'll look into the bug with save plot to workspace when a shape is not selected.

PR #26169 (2019-07-04): Fix save to workspace in instrument view
**Description of work.**

Fixes a bug reported by a user where the "Save to Workspace" functionality in both the miniplot and 3d view on the instrument view pick tab had stopped working. On workbench the latter caused a crash.

**Report to:** Pascal. ISIS.

**To test:**

1. Follow the instructions in the associated issue and see that a workspace now appears when requesting one from the miniplot.
2. To test that the plot works from the main 3D view then run the instrument view in workbench, and right click on a detector in the 3D view of the pick tab and [... cut]
closed 2019-07-04 by gvardany