mantidproject/mantid #26142: Latex error in Indirect testing documentation

LolloB, 2019-07-03:
The documentation for Indirect Inelastic Testing (http://developer.mantidproject.org/Testing/IndirectInelastic/IndirectInelasticAcceptanceTests.html) displays two Latex errors.

### Expected behavior
The page should display the plot expected as result of the testing procedure.

### Actual behavior
The file is not found, and an error returned

### Steps to reproduce the behavior
Go to page: _http://developer.mantidproject.org/Testing/IndirectInelastic/IndirectInelasticAcceptanceTests.html#data-analysis-elwin_

### Platforms affected
Tested on Firefox for Ubuntu 18

hsautessella, 2019-07-05:
I haven't been able to replicate the error on Windows 10 or my Ubuntu 16 or 18 VMs. The Ubuntu 16 VM is running the exact same version of pdfTex as is shown in the error message on the website. 

@martyngigg is there any config I might be missing when building locally?

martyngigg, 2019-07-05:
That's odd. It's built on a RHEL 7 machine so maybe it's something to do with an older version of Sphinx there.

Leave it with me. It's developer documentation and we have the source so it's not a priority for the release.

martyngigg, 2019-07-05:
I got that wrong. It's actually built on the Jenkins master node by the [developer_site](https://builds.mantidproject.org/job/developer_site/) job. My guess is that the latex installation is missing some package.

martyngigg, 2019-07-05:
I've installed the missing package and rebuilt the job. http://developer.mantidproject.org/Testing/IndirectInelastic/IndirectInelasticAcceptanceTests.html#data-analysis-elwin looks good now.

closed 2019-07-05 by martyngigg