mantidproject/mantid #26154: MantidPlot crashes when fitting

LolloB, 2019-07-03:
When using ConvFit for fitting a ElasticDiffRotDiscreteCircle function MantidPlot crashes without error messages.

### Expected behavior
The Interface performs the fit, if this is not possible it should return an error.

### Actual behavior
A segmentatuon fault occurs and MantidPlot crashes.

### Steps to reproduce the behavior
- Interaces->Indirect->Data analysis->ConvFit
- in single input load Sample and resolution file (eg. irs26176_graphite002_red.nxs and irs26173_graphite002_res.nxs)
- In Fit function->Fit Type select ElasticDiffRotDiscreteCircle

### Platforms affected
Tried on Ubuntu 18

robertapplin, 2019-07-04:
The code causing the problem is being replaced by Roman who is working on Simultaneous fitting, so this is low priority. Once simultaneous fitting has been finished, this issue should be checked again.

robertapplin, 2019-07-05:
@mantid-roman do you think this should be fixed for this release? If so, might it be possible for you to have a look?

mantid-roman, 2019-07-05:
I'll have a look

PR #26243 (2019-07-09): Indirect Data analysis interface fitting crash
**Description of work.**
Added a check on pointers before using them.

**To test:**

* Interaces->Indirect->Data analysis->ConvFit
* in single input load Sample and resolution file (eg. irs26176_graphite002_red.nxs and irs26173_graphite002_res.nxs)
* In Fit function->Fit Type select ElasticDiffRotDiscreteCircle
* No crashing
* Expand `Functions->f0-ElasticDiffRotDiscreteCircle` item. Check that `N`, `Q` and `WorkspaceIndex` properties have sensible values.
* Try other fit types.

Fixes #26154

---

#### Reviewer ####

Please comment on the followi [... cut]
closed 2019-07-09 by NickDraper