mantidproject/mantid #26157: Indirect Corrections crashes MantidPlot

LolloB, 2019-07-03:
MantidPlot crashes without warning if in the Indirect Corrections->Apply Absorption Corrections one sets the same file (or workspace) as input and Corrections

### Expected behavior
The interface should return an error

### Actual behavior
A segmentation fault happens and MantidPlot crashes

### Steps to reproduce the behavior
- Interfaces->Indirect->Corrections->Apply Absorption Correction
- Select a file (works with irs26174_graphite002_red in sample data)
- Add the file both in Sample and Corrections
- Press Run

### Platforms affected
Tested on Ubuntu 18

SpencerHowells, 2019-07-03:
Either the file must end _Corrections ie always restrict name or check that the file is a GroupWorkspace with the first member ending _ass

LolloB, 2019-07-04:
I just tried it on workbench and it crashes as well.

closed 2019-07-05 by robertapplin