mantidproject/mantid #26173: Indirect Bayes interface invalidates resolution workspace on MantidPlot

LolloB, 2019-07-04:
When running a fit with a range not overlapping with the resolution data the resolution workspace will be overwritten by other data. It will no longer work (even with a correct fitting range) until the Resolution file is reloaded.

### Expected behavior
An error is returned and the resolution workspace left unchanged.

### Actual behavior
The resolution workspace is overwritten necessitating to reload the file.

### Steps to reproduce the behavior
- Interfaces->Indirect->Bayes->ResNorm
- Load a Vanadium and resolution files (eg. irs26174_graphite002_red.nxs and irs26173_graphite002_res.nxs in sample data)
- Set EMin and EMax to cover a range not overlapping with the resolution data (eg. 0.4 and 0.5 respectively)
- Show data for resolution workspace, it should have 1 row
- Press Run, now the workspace will have multiple rows
- Return EMin and EMax to the original values
- Press Run again

### Platforms affected
Tested on Ubuntu 18

SpencerHowells, 2019-07-04:
The object of the program is to fit the shape of the peak. So why on earth would you chose a range which does not span the peak??

robertapplin, 2019-07-05:
Fixed in #26198

closed 2019-07-05 by robertapplin