mantidproject/mantid #26179: Indirect Symmetrise fails when EMax=0.4

LolloB, 2019-07-04:
When trying to symmetrise the data, the algorithm fails if the EMax value is in the range 0.399992 <= EMax <= 0.400090.

### Expected behavior
The data should be symmetrised correctly as it is with any other value of EMax.

### Actual behavior
The algorithm fails with error `Failed to calculate array slice boundaries: index -1 is out of bounds for axis 0 with size 0`

### Steps to reproduce the behavior
- MantidPlot->Interfaces->Indirect->Data Reduction->ISIS Energy transfer
- Set intrument=IRIS, Analyser=Graphite, Reflection=002
- Select a run number (eg. 26176) and press Run
- Data reduction->Symmetrise
- Input Workspace->iris<run number>_graphite002_red
- Set EMax to 0.4
- Press Preview

### Platforms affected
Tested on Ubuntu 18

robertapplin, 2019-07-04:
I can't reproduce this on windows. Perhaps someone on ubuntu could have a look a this

DTasev, 2019-07-09:
Reproduced on Ubuntu 18 as well, full debug log:
```
Added algorithm "Symmetrise" to batch queue
Starting next algorithm in queue: Symmetrise
Symmetrise started
Algorithm: Symmetrise v1
Execution Date: 2019-07-09 09:54:43.306732000
Execution Duration: -1 seconds
UUID: 88649bb2-e200-4c50-a6f5-93e46eac358b
Parameters:
  Name: InputWorkspace, Value: iris26176_graphite002_red, Default?: No, Direction: Input
  Name: SpectraRange, Value: , Default?: Yes, Direction: Input
  Name: XMin, Value: 0.054325428563315702, Default?: No, Direction: Input
  Name: XMax, Value: 0.40000000000000002, Default?: No, Direction: Input
  Name: OutputWorkspace, Value: iris26176_graphite002_sym_red, Default?: No, Direction: Output
  Name: OutputPropertiesTable, Value: __SymmetriseProps_temp, Default?: No, Direction: Output
Read-locking iris26176_graphite002_red
CropWorkspace started (child)
Algorithm: CropWorkspace v1
Execution Date: 2019-07-09 09:54:43.312262000
Execution Duration: -1 seconds
UUID: dffad43a-05f7-4b28-8f45-709d0ee1853e
Parameters:
  Name: InputWorkspace, Value: iris26176_graphite002_red, Default?: No, Direction: Input
  Name: OutputWorkspace, Value: __symm, Default?: No, Direction: Output
  Name: XMin, Value: 8.9884656743115785e+307, Default?: Yes, Direction: Input
  Name: XMax, Value: 8.9884656743115785e+307, Default?: Yes, Direction: Input
  Name: StartWorkspaceIndex, Value: 0, Default?: Yes, Direction: Input
  Name: EndWorkspaceIndex, Value: 50, Default?: No, Direction: Input
Time to validate properties: 0.000013 seconds
Time for other input validation: 0.000001 seconds
Time for other initialization: 0.000031 seconds
Time to run exec: 0.001379 seconds

ExtractSpectra finished with isChild = 1
Data Object '__symm' replaced in data service.

Time to validate properties: 0.000016 seconds
Time for other input validation: 0.000002 seconds
Time for other initialization: 0.000744 seconds
Time to run exec: 0.002077 seconds

CropWorkspace successful, Duration 0.00 seconds
Unlocking iris26176_graphite002_red
Error in execution of algorithm Symmetrise
Failed to calculate array slice boundaries: index -1 is out of bounds for axis 0 with size 0
  at line 87 in '/home/dtasev/dev/m/Framework/PythonInterface/plugins/algorithms/Symmetrise.py'
Got error from algorithm "Symmetrise"
Stopping batch algorithm because of execution error
```

DTasev, 2019-07-09:
This workspace can be used to reproduce the Symmetrise error: 
[iris26176_graphite002_red.zip](https://github.com/mantidproject/mantid/files/3372452/iris26176_graphite002_red.zip)

alicerussell1, 2019-07-11:
I can replicate this on windows with this: 
[iris26176_graphite002_red.zip](https://github.com/mantidproject/mantid/files/3381141/iris26176_graphite002_red.zip)

PR #26298 (2019-07-11): Symmterise algorithm: New implementation for finding XMax
**Description of work.**
Changes the logic used to find XMax. The -XMin and +XMin logic is unchanged.

**To test:**
Follow instructions from associated issue. Unit tests should pass

Fixes #26179

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/IndividualTicketTesting/)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the code conform to the [coding standards](http://developer.mantidproject.org/Standards/)?
- Are the unit tests small and test the c [... cut]
closed 2019-07-11 by NickDraper