mantidproject/mantid #26183: Calculate Paalman Pings crashes the workbench

LolloB, 2019-07-04:
In the calculate Paalman Pings tab, when loading a file that ends in _sqw selecting a chemical formula and running the algorithm, the workbench crashes  due to an unhandled exception:
`what(): Invalid value for property InputWorkspace (MatrixWorkspace) from string "iris26176_graphite002_iqt": The workspace must have units`

### Expected behavior
An error message should be displayed if the input file is invalid.

### Actual behavior
The algorithm fails and the workbench crashes.

### Steps to reproduce the behavior
- workbench->Interfaces->Indirect->Corrections->Calculate Paalman Pings
- As input select a _sqw file (eg. ExternalData/Testing/Data/SystemTest/BASIS_59689_divided_sqw.nxs)
- Input a chemical formula (eg. H2-O)
- Click Run

### Platforms affected
Tested on workbench for Ubuntu 18

SpencerHowells, 2019-07-04:
A _iqt file is illegal! The filter should be to only access _red files.

LolloB, 2019-07-04:
The filter also show _sqw files which also crash the workbench with exception `what(): Instrument has no efixed parameter`

SpencerHowells, 2019-07-04:
BASIS is a SNS instrument. I don't know how they reduce their data to sqw but it seems that they do not set a value for efixed in the workspace. This is a basic Mantid parameter required to convert time-of-flight to energy transfer and is used by ISIS & ILL instruments.

PR #26198 (2019-07-05): Indirect Fix Corrections Crashes
**Description of work.**
This PR fixes two bugs (to save some time):
- A crash on CalculatePaalmanPings when Efixed cannot be found for an instrument.
- A crash when attempting to use an incorrect Corrections workspace on ApplyAbsorptionCorrections.

**To test:**
1.`Interfaces`->`Indirect`->`Data Corrections`->`CalculatePaalmanPings`
See instructions in #26183 
A message should appear asking the user to enter an Efixed value.

1.`Interfaces`->`Indirect`->`Data Corrections`->`ApplyAbsorptionCorrections`
See instructions in #26157 
A message should appear asking  [... cut]
closed 2019-07-05 by NickDraper