mantidproject/mantid #26185: Mslice - failed to load .txt files

alicerussell1, 2019-07-04:
### Expected behavior
If you save a file using the save button in the workspace manager in a `.txt` format and try and load it in the data loading tab, the workspace should load 

### Actual behavior
It returns the following error
```
Traceback (most recent call last):
  File "C:\MantidNightlyInstall\scripts\ExternalInterfaces\mslice\widgets\dataloader\dataloader.py", line 55, in activated
    self.load(False)
  File "C:\MantidNightlyInstall\scripts\ExternalInterfaces\mslice\widgets\dataloader\dataloader.py", line 104, in load
    self._presenter.load_workspace(self.get_selected_file_paths(), merge)
  File "C:\MantidNightlyInstall\scripts\ExternalInterfaces\mslice\presenters\data_loader_presenter.py", line 34, in load_workspace
    self._load_ws(file_paths, ws_names)
  File "C:\MantidNightlyInstall\scripts\ExternalInterfaces\mslice\presenters\data_loader_presenter.py", line 47, in _load_ws
    load_from_ascii(file_paths[i], ws_name)
  File "C:\MantidNightlyInstall\scripts\ExternalInterfaces\mslice\models\workspacemanager\file_io.py", line 140, in load_from_ascii
    cut_axis = Axis(*cut_axis_str[:cut_axis_str.find('\n')].split(','))
TypeError: __init__() takes at least 5 arguments (2 given)
```

martyngigg, 2019-07-04:
Thanks for spotting this. I've copied the issue the [mslice](/mantidproject/mslice) repository too.

alicerussell1, 2019-07-08:
Fixed in [#477](https://github.com/mantidproject/mslice/pull/477)

closed 2019-07-08 by alicerussell1