mantidproject/mantid #26188: Indirect diffraction initializes to the wrong number of spectra

LolloB, 2019-07-04:
When analyzing IRIS data, the automatic initialization of Spectra Min and Max will set them to a non-existing number of spectra, thus requiring manual adjustment each time.

### Expected behavior
Like for the other instruments, the correct range of spectra should be detected and set.

### Actual behavior
Loading the run will set Spectra Min and Max to values outside the actual range of spectra.

### Steps to reproduce the behavior
- Interfaces->Indirect->Diffraction
- Set Instrument=IRIS, Analyser=diffraction, reflection=diffspec
- Load a run (eg. 200)
- For run 200 the real range of spectra is 1-5, the values will however be initialized to 105-112

### Platforms affected
Tested on Ubuntu 18

SpencerHowells, 2019-07-04:
Run 200 is very very old ~20 yrs - during the building of the instrument. It is utterly stupid to use it.

robertapplin, 2019-07-04:
I think this is too low priority (if a priority at all), so I will close this.

closed 2019-07-04 by robertapplin