mantidproject/mantid #26195: Loading large numbers of runs in Muon Analysis

Mantid-Matthew, 2019-07-04:
If you attempt to load more than 100 runs in the Muon Analysis GUI an exception is thrown saying that it can only load a maximum of 100 at any one time. This exception is not caught and makes it to the last chance error handler.

This needs to be investigated and either the limit should be removed or the exception should be handed more gracefully.

LolloB, 2019-07-05:
https://github.com/mantidproject/mantid/blob/master/scripts/Muon/GUI/Common/utilities/run_string_utils.py#L17

LolloB, 2019-07-05:
When loading runs 62022-122 the first check in run_string_to_list() passes but another one fails later only after all the runs have loaded, a popup appears and an error returned.

PR #26220 (2019-07-15): Removed limit on number of runs loaded by Muon Analysis 2
**Description of work.**
Removed the limit on the maximum number of runs that can be loaded at one time in the `Data Analysis 2` interface.

**To test:**
- `Interfaces->Muon->Muon Analysis 2->Home`
- Write the number of the runs to be loaded, eg. `62022-122`, the input fields should be grayed out, the data loaded and plotted.
Note that loading large number of runs (more that ~100) will require several minutes.

Fixes #26195.

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ [... cut]
closed 2019-07-15 by NickDraper