mantidproject/mantid #26208: LoadDialog() cannot be aborted if closed with "x" button

ConorMFinn, 2019-07-05:
### Expected behavior
After running the LoadDialog() function from the python window, clicking the red x should abort the algorithm input.
### Actual behavior
If the run button has been mistakenly clicked before selecting a file, or selecting an invalid file path, clicking the red x continues execution of the script until the whole program is forced closed. 
Note: Using the close button still cancels the execution. 
### Steps to reproduce the behavior
1. Open the Python Window (F3)
2. Type `LoadDialog()`
3. Run the script.
4. Put nothing or an invalid file path in the File field. 
5. Press enter or click Run.
6. Close the dialog with the red x
7. The python window is now unusable and will not abort until the program is closed. 

### Platforms affected
Tested on Ubuntu 18.04

martyngigg, 2019-07-17:
I think the "fix" here should be to remove the `*Dialog` functions from the Python API (maybe just for workbench?). They have never been used seriously by users and at this point they are just causing confusion.

martyngigg, 2021-02-23:
A duplicate of the fixed issue #28669

closed 2021-02-23 by martyngigg