mantidproject/mantid #26217: Move to VS2019

NickDraper, 2019-07-05:
1. Check that mantid will compile in VS2019
1. Create a simple instruction list for everyone else to follow
1. Organise a time for everyone to migrate

hsautessella, 2019-07-30:
# Migrating from Visual Studio 2017 to 2019

1) Download the latest version of [CMake (v3.15)](https://cmake.org/download/).

2) Download the latest [Visual Studio installer](https://visualstudio.microsoft.com/vs/community/)

3) Launch the installer and install Visual Studio 2019. Make sure the following components are installed:

  - Desktop development with C++
  - The latest Windows 10 SDK
  - Windows Universal CRT SDK

  As a reference, below are the components I have installed:

[image]
  
  You will need to restart your computer after installation.

(*If you're not overwriting an existing VS 2017 build you can ignore steps 4 and 5.*)

4) Delete the `CMakeFiles` folder and `CMakeCache.txt` from the following directories:

  - `<build-directory>/`
  - `<build-directory>/extern-span/`
  - `<build-directory>/googletest-download/`

5) Open the CMake GUI and click `File → Delete Cache`.

6) Configure CMake and select `Visual Studio 16 2019` as the generator. Then generate.

7) Now open Visual Studio 19 using the `visual-studio.bat` file in the build directory.

8) Build then go and get a coffee.

There is a known issue of external library warnings (particularly from Qt4) being raised by Intellisense. Unfortunately I haven't found a good way of telling Intellisense to ignore headers from the third party directory (if you can, let me know!). You can disable the warnings for specific targets by selecting the target then opening `View → Property Pages → Advanced → Disable Specific Warnings` and enter the third party directory include directory (`...\mantid\external\src\ThirdParty\include`). Unfortunately re-running CMake recreates the target and the setting is lost.

If these warnings bother you too much you can disable Intellisense errors by going to `Tools → Options → Text Editor → C/C++ → Advanced → Intellisense` and setting `Disable Error Reporting` to `True`. Alternatively, in the `Error List` window, you can show errors/warnings from "Build Only", rather than "Build + Intellisense" (see below).

[image] → [image]

martyngigg, 2019-08-08:
With #26562 merged this is now complete.

closed 2019-08-08 by martyngigg