mantidproject/mantid #26235: Sliceviewer crashes on windows with Matrix workspaces

NickDraper, 2019-07-08:
### Actual behavior
The sliceviewer crashes the workbench when trying to open the sliceviewer on a "normal" matrix workspace on windows.

It seems to start ok for MDWorkspaces. and it seems to be ok on linux.

### Steps to reproduce the behavior
1. open workbench
1. Load a worspace
1. right click and select sliceviewer
1. The window will appear white for a couple of seconds and then crash out to the error reporter.

The following file definately crashes, but it seems to be consistent for any matrix workspace.
SXD23767.raw (can be found in the training data)

### Platforms affected
Windows 8 and 10, although linux seems to be ok.

NickDraper, 2019-07-08:
@rosswhitfield Ross, I've asked Harry to look at this first as it seems to be only on windows, but if you could help at all it would be appreciated.

hsautessella, 2019-07-08:
This does also appear to be a problem on the master build of my Ubuntu 18 VM. 

The line causing the issue seems to be: https://github.com/mantidproject/mantid/blob/6e380560c08100841365b1e1d93d0945683d22cd/qt/python/mantidqt/widgets/sliceviewer/samplingimage.py#L59 

I'll carry on looking into it, but just giving a heads up that it's also happening on Ubuntu 18

martyngigg, 2019-07-08:
I assume this is to do with either numpy or matplotlib versions which is why probably I can't see it.

PR #26255 (2019-07-12): Fix sliceviewer crash 
### **This should be tested by someone on Windows or Ubuntu 18**

**Description of work.**
A crash when opening Slice Viewer was being caused by a C++ `vector` being de-referenced before being passed through the `boost::python` API, this was causing a segfault. This only seemed to be occuring on Windows and Ubuntu 18.

The python API function `MatrixWorkspace::getSignalCoord` has been re-written to create a numpy array on the C++ side rather than a vector and pass this directly back to the Python. This also avoids copying a vector into a numpy array.

**To test:**
- Runn [... cut]
closed 2019-07-12 by NickDraper