mantidproject/mantid #26250: ALC crash

AnthonyLim23, 2019-07-09:
Open workbench
Set default instrument to EMU
Open ALC
For first enter 94183
for Last enter 94185
Click Load
Go to baseline modelling
Add a fitting function and region
Do a fit
Go to peak fitting and do a fit
Go back to data loading
Then for first and last browse to some HIFI data
Click Load
Mantid will load the data and then through an error.

AnthonyLim23, 2019-07-09:
this was not present in v 4.0

Pasarus, 2019-07-09:
This is the error I am getting:
[image]

and if I load again immediately afterwards it causes a segfault

robertapplin, 2019-07-09:
I have found the fix for this @Pasarus . Are you ok with me creating a PR to fix this?

Pasarus, 2019-07-09:
@robertapplin If you are working on an issue on the project board could you assign yourself next time? also it would be great so I don't need to continue.

AnthonyLim23, 2019-07-09:
thats the bug

PR #26256 (2019-07-11): Muon Fix crash when Selector is removed
**Description of work.**
This PR fixes a crash caused by attempting to remove a marker which has already been removed from the patch.

No release notes required as this is only present for workbench, and ALC wasn't in the workbench for the last release.

**To test:**
Open workbench
Set default instrument to EMU
Open ALC
For first enter 94183
for Last enter 94185
Click Load
Go to baseline modelling
Add a fitting function and region
Do a fit
Go to peak fitting and do a fit
Go back to data loading
Then for first and last browse to some HIFI data
Click Load
 [... cut]
closed 2019-07-11 by martyngigg