mantidproject/mantid #26254: Algorithm Progress in workbench not automaticaly showing live data loading

Harrietbrown, 2019-07-09:
### Expected behavior
When loading live data in workbenchthe process should alway appear on the Algorithm progress window while it is running
### Actual behavior
If you open the algorithm progress dialog box while loading live data it is initialy empty until the next time the live data updates.
### Steps to reproduce the behavior
1. Open Workbench.
1. In the workspaces widget ``load`` --> ``live data``
1. Set to update every 3 seconds and run live data monitoring.
1. In the algorithm widget open details.
1. The algorithm list should be empty until the next time the data updates.
1. Close and reopen details. the list should be empty again until the next update. 
### Platforms affected
Windows

DTasev, 2019-07-09:
This is due to algorithms only being added on `AlgorithmProgressModel::startHandle` and `AlgorithmProgressModel::progressHandle`. Running algorithms `Details` are not currently shown, until their report progress. This should be added to the constructor of the `AlgorithmProgressDialogPresenter`, maybe via expanding the model to have a `getAllRunningAlgorithms` function.

Additionally the main progress bar should show `Live Data running` while the Live Data algorithm is in progress

NickDraper, 2020-01-31:
duplicate of #27077

closed 2020-01-31 by NickDraper