mantidproject/mantid #26274: Indirect Settings - Extra settings options

robertapplin, 2019-07-10:
### Description
The Indirect Settings GUI should be re-designed so there are two sections: 
- An Input Data section: this sections deals with settings related to the input data you want to load.
- A Plot Options section: this sections allows you to customize your plots

**Input Data Section**
- [X] Restrict Input Data By Name
- [] Load History
- [] Load logs

**Plot Options Section**
- [X] Plot error bars
- [] Error bar styles (matplotlib allows a large number of customize-able properties)
- [] Line styles

### Platforms affected
All

AnthonyLim23, 2019-08-14:
are these mini-plots?

robertapplin, 2019-08-15:
This is referring to the external plotting, (so not the mini-plots within the interface). External plots are usually triggered using the `Plot Current Preview` button (which plots the spectra currently visible in the mini-plots), and using the output plotting options.

closed 2021-03-02 by stale[bot]