mantidproject/mantid #26301: Workbench Windows DPI scaling

hsautessella, 2019-07-11:
### Expected behavior
Workbench's display should scale on a per-monitor basis, including when remoting into another PC. 

### Actual behavior
#26266 set Workbench's DPI awareness such that Windows scales the view to a resolution that is suitable for all connected monitors. Windows will essentially zoom in and out on the view when the window is moved. Whilst this scales the view sensibly it's not ideal as it can cause blurriness.

This issue should attempt to set the DPI awareness to 2 (i.e. per-monitor) and have the workbench scale it's widgets and fonts sensibly whilst maintaining its sharpness. This may involve updating Scintilla as we are currently running v2.10 and some bug fixes for DPI scaling were implemented in v3.6.5 (https://www.scintilla.org/ScintillaHistory.html).

Some useful links:
https://doc.qt.io/qt-5/highdpi.html#high-dpi-displays
https://stackoverflow.com/questions/39823918/how-to-approach-development-of-new-qt-5-7-high-dpi-per-monitor-dpi-aware-applic

### Steps to reproduce the behavior
- Open Workbench on a Windows PC connected to 2 monitors with different resolutions. 
- You should observe some difference in sharpness of the Workbench's view when moving the window between monitors.

### Platforms affected
Windows

NickDraper, 2019-07-22:
Next task is a timeboxed investigation

hsautessella, 2019-08-23:
# DPI Investigation

## The following is the same as setting dpiawareness=1 in the qt.conf:

```python
import ctypes
user32 = ctypes.windll.user32.SetProcessDPIAware()
```

4k and 2k screen respectively:

[image] [image]

## Now setting dpiawareness=2 and setting:
The DPI awareness can be set on line 25 of `mantid/qt/applications/workbench/CMakeLists.txt`. Since 2 is the default, you can also just remove the section that's currently setting it to 1.

This needs to be done before the QApp is created:
```python
QCoreApplication.setAttribute(Qt.AA_EnableHighDpiScaling)
```
or the following can be done before importing *any* Qt modules:
```python
os.environ['QT_SCALE_FACTOR'] = "1"
os.environ['QT_AUTO_SCREEN_SCALE_FACTOR'] = "1"
os.environ['QT_SCREEN_SCALE_FACTORS'] = "2"  # this can be set on a per monitor basis with semicolon delimited list
```

4k and 2k screen respectively:

[image] [image]

The 4k screen looks the same with both settings. The text in the 2k screen is significantly smaller when high dpi scaling is enabled.

## Setting the scale factor on a per monitor basis:
```python
os.environ['QT_AUTO_SCREEN_SCALE_FACTOR'] = "1"
os.environ['QT_SCREEN_SCALE_FACTORS'] = "1;1.5"
```
or
```python
os.environ['QT_SCALE_FACTOR'] = "1.5"
os.environ['QT_AUTO_SCREEN_SCALE_FACTOR'] = "1"
```

[image] [image]

The sizing in this case is perfect, however Qt does not play nicely with non-integer scale factors
(it rounds them).
This is why the text is distorted on the 2k image. We could also have set the scale factor list to
`"0.6667;1"` and this time the 4k screen would be similarly distorted.

We would have to do some screen scale-factor calculations to get the relative size of the screens before setting these environment variables. The issue being that we wouldn't be able to adapt the scaling if the user changed their scree resolution mid-session.

## Updating QScintilla
Updating QScintilla from `2.10.1` -> `2.11.2` solved the rendering issues inside the script window.

The best solution so far - for my dual monitor, 4k (scaled up by 150%) and 2k setup - is:

    - Update QScintilla to version 2.11.2
    - Before importing Qt, set the enivoronment variables:
        os.environ['QT_AUTO_SCREEN_SCALE_FACTOR'] = "1"
        os.environ['QT_SCREEN_SCALE_FACTORS'] = "1.5;1"

This scales text and buttons correctly on both screens, with no blurry text inside the script editor
as before.

There are some visual artefacts however. As an example here's the "Manage User Directories" window (4k on the left):
 
[image] [image]
[image]

The borders of some buttons/text boxes are cut-off.

## Conclusions
Setting `Qt.AA_EnableHighDpiScaling` sets the correct resolutions on each monitor, however the scaling is not quite right, and the text on my lower resolution monitor was too small.

Setting relative scale factors between the two screens scales the GUI - including text - perfectly but leaves some visual artefacts. 

QScintilla should be updated as this seems to fix bluriness issues within the script editor.

martyngigg, 2019-09-02:
We'll update QScintilla and leave it at that for now.

closed 2019-09-02 by martyngigg