mantidproject/mantid #26333: Workbench Fit Parameters are not updating, and no Fit is plotted

robertapplin, 2019-07-15:
### Expected behavior
The fits for the spectra should be plotted, and the parameters in the FitPropertyBrowser should be updated when changing between the workspace indices.

### Actual behavior
No fit is plotted. The parameters are not updated.

### Steps to reproduce the behavior
1. Workbench
2. Load the data below
3. Right click the workspace and plot the first index
4. Click `Fit`
5. Select `Lorentzian` and `Flat Background`
6. `Fit`->`Sequential Fit`->`Ok`
No fits plotted, and parameters won't update.

**Data**
[irs26176_graphite002_red.zip](https://github.com/mantidproject/mantid/files/3391843/irs26176_graphite002_red.zip)

### Platforms affected
All

robertapplin, 2019-07-15:
Occasionally we see that 2 spectra are plotted after doing a Fit. I think these are the fit curve and difference for the selected spectrum.

alicerussell1, 2019-07-17:
After looking into this I have found that the sequential fit dialog depends on the mantidplot UI so the signals are not being passed to the fit browser. When you load the dialog box in workbench it gives the message `QObject::connect: Cannot connect MantidQt::MantidWidgets::SequentialFitDialog::needShowPlot(Ui::SequentialFitDialog *, MantidQt::MantidWidgets::FitPropertyBrowser *) to (null)::showSequentialPlot(Ui::SequentialFitDialog *, MantidQt::MantidWidgets::FitPropertyBrowser *)`.

martyngigg, 2019-07-18:
I have added a [release note](https://github.com/mantidproject/mantid/tree/release-next) to the workbench in 4.1 to state that this is a known issue.

PR #27108 (2019-11-01): Fix sequential fitting bug to update parameters and do plot
**Description of work.**
This PR fixes the bug in sequential fitting so that when a sequential fit is done the function parameters are updated and a plot is done.

**To test:**
Follow instructions in #26333 to do a sequential fit. 
The parameters should update to match those in the workspace.
A plot of this function should also be done.

Fixes #26333 

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of  [... cut]
closed 2019-11-05 by alicerussell1