mantidproject/mantid #26384: Pre-processing bad data in Engineering diffraction causes a hard crash

LolloB, 2019-07-18:
### Expected behavior
A warning should be returned and the user allowed to change file.

### Actual behavior
MantidPlot crashes and the following error message is returned:
> ********* UNHANDLED EXCEPTION *********
  what(): std::bad_alloc
Backtrace:
 /home/lorenzo/mantid-build/bin/libMantidAPI.so(+0x35d857) [0x7f34537a5857]
 /home/lorenzo/mantid-build/bin/libMantidAPI.so(+0x35d97b) [0x7f34537a597b]
 /usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x92ab6) [0x7f344c72bab6]
 /usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x92af1) [0x7f344c72baf1]
 /usr/lib/x86_64-linux-gnu/libstdc++.so.6(__cxa_rethrow+0x49) [0x7f344c72bd79]
 /usr/lib/x86_64-linux-gnu/libQtCore.so.4(_ZN11QMetaObject8activateEP7QObjectPKS_iPPv+0xcd0) [0x7f344f490770]
 /usr/lib/x86_64-linux-gnu/libQtCore.so.4(+0x7ae25) [0x7f344f36be25]
 /lib/x86_64-linux-gnu/libpthread.so.0(+0x76db) [0x7f344bc9c6db]
 /lib/x86_64-linux-gnu/libc.so.6(clone+0x3f) [0x7f344b9c588f]

### Steps to reproduce the behavior
1. `interfaces->Diffraction->Engineering Diffraction->Pre-processing`
2. Load a run that doesn't need pre-processing, eg. `ExternalData/Testing/Data/SystemTest/ENGINX00193749.nxs`
3. Press `Rebin`

### Platforms affected
Tested on Ubuntu 18.04

DTasev, 2019-07-18:
Interesting, I can't replicate this also on Ubuntu 18.04. Could you right click on the Result Log window, set the level to Debug, and run it again. If you could attach anything it prints out that seem relevant it might make it easier to find the cause

LolloB, 2019-07-19:
>Load-[Notice] Load successful, Duration 0.60 seconds
Rebin-[Notice] Rebin started
Rebin-[Information] Algorithm: Rebin v1
Rebin-[Information] Execution Date: 2019-07-19 15:05:47.030675000
Rebin-[Information] Execution Duration: -1 seconds
Rebin-[Information] UUID: ca6bcac4-4a89-4327-8254-5c490e04c4ec
Rebin-[Information] Parameters:
Rebin-[Information]   Name: InputWorkspace, Value: engggui_preproc_input_ws, Default?: No, Direction: Input
Rebin-[Information]   Name: OutputWorkspace, Value: engggui_preproc_time_ws, Default?: No, Direction: Output
Rebin-[Information]   Name: Params, Value: 1e-05, Default?: No, Direction: Input
Rebin-[Information]   Name: PreserveEvents, Value: 1, Default?: Yes, Direction: Input
Rebin-[Information]   Name: FullBinsOnly, Value: 0, Default?: Yes, Direction: Input
Rebin-[Information]   Name: IgnoreBinErrors, Value: 0, Default?: Yes, Direction: Input
Rebin-[Debug] Read-locking engggui_preproc_input_ws
Rebin-[Information] Using the current min and max as default 6000, 45999
Rebin-[Error] Error in execution of algorithm Rebin:
Rebin-[Error] std::bad_alloc
Rebin-[Debug] Unlocking engggui_preproc_input_ws

>********* UNHANDLED EXCEPTION *********
  what(): std::bad_alloc

>Backtrace:
 /opt/Mantid/bin/../lib/libMantidAPI.so(+0x1a1fb9) [0x7fe751234fb9]
 /opt/Mantid/bin/../lib/libMantidAPI.so(+0x1a2160) [0x7fe751235160]
 /usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x92ab6) [0x7fe749cfaab6]
 /usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x92af1) [0x7fe749cfaaf1]
 /usr/lib/x86_64-linux-gnu/libstdc++.so.6(__cxa_rethrow+0x49) [0x7fe749cfad79]
 /usr/lib/x86_64-linux-gnu/libQtCore.so.4(_ZN11QMetaObject8activateEP7QObjectPKS_iPPv+0xcd0) [0x7fe74ab02770]
 /usr/lib/x86_64-linux-gnu/libQtCore.so.4(+0x7ae25) [0x7fe74a9dde25]
 /lib/x86_64-linux-gnu/libpthread.so.0(+0x76db) [0x7fe74a3966db]
 /lib/x86_64-linux-gnu/libc.so.6(clone+0x3f) [0x7fe74955188f]
QSocketNotifier: Invalid socket 17 and type 'Read', disabling...
QThreadStorage: Thread 0x557e0cebecd0 exited after QThreadStorage 3 destroyed
QThreadStorage: Thread 0x557e0cec79a0 exited after QThreadStorage 3 destroyed

Tested on the release candidate

Pasarus, 2019-07-22:
On Ubuntu 19.04 I could not replicate this

closed 2021-03-02 by stale[bot]