mantidproject/mantid #26388: General fourier transform algorithm

mantid-roman, 2019-07-18:
**Original reporter:** Spencer Howells

Create a general Fourier transform algorithm that will work for any binning (FFT only works for regular binning).

AnthonyLim23, 2021-03-01:
I think we might need this

martyngigg, 2021-03-04:
Do we have a concrete requirement for this or is this just a nice to have. If it's concrete requirement I can probably get this scheduled for Core soon but I don't really want to do that if there is no real need.

SpencerHowells, 2021-03-04:
I think we had 1 user who asked for it. It was a bit of an odd request.

martyngigg, 2021-03-08:
If it's 1 user and we think it's not too useful then I think we need to seriously think about it and close it if we really don't intend to do it.

If a fresh requirement pops up it can always be reopened or a new task defined.

AnthonyLim23, 2021-04-15:
I am seeing complaints from users about FFT's producing peaks with wider peaks than they would expect.

martyngigg, 2021-04-16:
Is this a bug with the current algorithms then? If so can we open bug reports for those?

AnthonyLim23, 2021-04-16:
At this point its not clear. It is possible that some of the assumptions of a fast Fourier Transform mean that the result is different from what the user expects.

AnthonyLim23, 2021-04-16:
it does look like a bug in FFT. I have opened an issue https://github.com/mantidproject/mantid/issues/31150

martyngigg, 2021-06-23:
Closing as the issue with FFT has been resolved in the referenced issue above.

closed 2021-06-23 by martyngigg