mantidproject/mantid #26393: Peak selector lines in Indirect Bayes move only in one direction

LolloB, 2019-07-18:
### Expected behavior
Peak selector lines should be dragged in both direction like in the other interfaces.

### Actual behavior
The `EMin` line only moves to the right, the `EMax` only to the left. When they reach the same position they get "stuck" and they can only be moved back by resetting the numerical value.

### Steps to reproduce the behavior
1. `Interfaces->Indirect Bayes->ResNorm`
2. Load some data for vanadium and resolution (eg. rs26173_graphite002_red.nxs, irs26173_graphite002_res.nxs)
3. Drag the peak selector lines

### Platforms affected
Tested on Ubuntu 18.04

AnthonyLim23, 2019-07-19:
is anyone currently working on this?

NickDraper, 2019-07-19:
All works fine now, even crossing the lines

PR #26427 (2019-07-19): Only set min/max bounds for range selector if they are provided.
**Description of work.**

Fixes range selector being clamped when dragged on Indirect ResNorm interface.

**To test:**

Follow the instructions in the issue. 

Fixes #26393.

*This does not require release notes* because **it is a regression in this release.**

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/IndividualTicketTesting/)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the code conform to the [coding standards](http://developer. [... cut]
closed 2019-07-19 by NickDraper