mantidproject/mantid #26396: Closing Indirect interfaces causes mantid to crash

LolloB, 2019-07-18:
When `Indirect Bayes` or `Indirect Corrections` are closed immediately after loading a file, MantidPlot will crash with the error message:
> Segmentation fault (core dumped)

If something is done (eg. a button pressed, tab changed etc.) before closing then the crash does not occur.

### Expected behavior
The window should close successfully.

### Actual behavior
MantidPlot crashes.

### Steps to reproduce the behavior
1. `Interfaces->Indirect->Bayes` (or `Corrections`)
2. Load some data (eg. irs26173_graphite002_red, irs26173_graphite002_res)
3. Without touching anything else, close the window

### Platforms affected
Tested on Ubuntu 18.04

AnthonyLim23, 2019-07-19:
I can't get this on windows. @Mantid-Matthew can you reproduce this?

DTasev, 2019-07-19:
I can't replicate this on Ubuntu 18 compiled at eca4d2834cc

NickDraper, 2019-07-19:
This works fine for me on Windows 7 with the release candidate, so I'm going to close this one for now.

closed 2019-07-19 by NickDraper