mantidproject/mantid #26406: SANS Documentation: SANSStitch

Pasarus, 2019-07-19:
### Expected behavior
No warnings on building `docs-html`

### Actual behavior
Get this warning:
```
reading sources... [ 46%] algorithms/SANSSolidAngleCorrection-v1
reading sources... [ 46%] algorithms/SANSStitch-v1
Warning: Unable to reclaim box space in spline routing for edge "subtractCanLAB" -> "valQLAB". Something is probably seriously wrong.
reading sources... [ 46%] algorithms/SANSSubtract-v1
reading sources... [ 46%] algorithms/SANSWideAngleCorrection-v1
```

### Steps to reproduce the behavior
Do a clean `docs-html` target build

### Platforms affected
Ubuntu

DavidFair, 2019-11-11:
This is still an issue, looking at it we need to tidy down the graph as DOT is having problems routing the line of this edge

Pasarus, 2021-02-23:
@gemmaguest @DavidFair Is this issue still a problem if not I would like to close.

DavidFair, 2021-02-23:
I've not run into it since and everyone forgot about it, so lets stale it

closed 2021-02-23 by DavidFair