mantidproject/mantid #26408: Clean/Dirty Install Smoke Tests

NickDraper, 2019-07-19:
# Clean/dirty install smoke tests

Please run these tests on the compiled package of Mantid; **not a locally compiled version**.

When reporting back please include:

- Your OS
- Your facility/location
- If you have other package versions of Mantid installed (yes/no)

If you have any questions please don't hesitate to contact the creator of this issue.

## Checklist/directions

### Dirty install

* Make sure that you have several versions of Mantid installed
	* Last release
	* A nightly
	* If possible an old release
* Install the latest version of the new Mantid
- [ ] Check that Mantid boots up correctly

### Clean install

* Remove all existing Mantid versions and associated files
* Windows:
	* Uninstall the program
	* Clear shortcuts from desktop
	* Clean out the registry
		* Load regedit (Command Prompt > regedit)
		* Find `HKEY_CURRENT_USER > SOFTWARE > Mantid` delete it
* OSX:
	* Remove the application
	* Remove the `~/.mantid directory`
	* Remove `~/Library/Preferences/org.mantidproject.MantidPlot.plist`
* Linux
	* Remove the package: `/opt/Mantid`
	* Remove `~/.config/Mantid`
	* Remove `~/.mantid/`

* Re-install the latest version of the new Mantid
- [ ] Check that Mantid boots up correctly

DTasev, 2019-07-19:
Testing on Ubuntu 16.04, 18.04 and CentOS7

Pasarus, 2019-07-22:
Windows 10 - all fine.

martyngigg, 2019-07-23:
Tested on OSX 10.12 and all seems fine.

martyngigg, 2019-07-23:
Tested on RHEL 7 and all seems fine.

closed 2019-07-23 by NickDraper