mantidproject/mantid #26410: Fitting Smoke Tests

NickDraper, 2019-07-19:
# Fitting smoke tests

Please run these tests on the compiled package of Mantid; **not a locally compiled version**.

The check boxes should correspond to headings in the documentation (linked). Just follow the corresponding directions.

When reporting back please include:

- Your OS
- Your facility/location
- If you have other package versions of Mantid installed (yes/no)

If you have any questions please don't hesitate to contact the creator of this issue.

## Checklist/directions

* Fitting to data [directions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/01_fitting_models_to_data.html#fitting-models-to-data)
	* Simple fit [ ]
	* Activating the fitting tools [ ]
	* Selecting the fitting range [ ]
	* Other Fit Function Settings properties [ ]
	* Setting up a fit model [ ]
	* Adjusting fit function parameter [ ]
	* Saving a model [ ]
	* Tying and constraining fit parameters [ ]
	* Execute your fit [ ]
	* Inspecting a fit result [ ]
* Additional options [directions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/02_fit_model_choices.html#fit-model-choices)
	* Non peak model + background [ ]
	* User defined function [ ]

giovannidisiena, 2019-07-19:
Tested on Ubuntu 18.04 VM with no issues.  Other package versions of Mantid installed (yes)

ciaranightingale, 2019-07-19:
Windows 10, ISIS, no issues found.

giovannidisiena, 2019-07-22:
**All Tests Passed | Plot and Workbench**
**OS:** Ubuntu 18.04
**Facility:** ISIS
**Other Versions:** Yes

hsautessella, 2019-07-22:
##### Windows 10, ISIS. Installed alongside a Nightly and an Unstable build.

No major issues found on MantidPlot or Workbench.

closed 2019-07-22 by NickDraper