mantidproject/mantid #26411: Interface smoke tests

NickDraper, 2019-07-19:
# Interface/misc smoke tests

Please run these tests on the compiled package of Mantid; **not a locally compiled version**.

When reporting back please include:

- Your OS
- Your facility/location
- If you have other package versions of Mantid installed (yes/no)

If you have any questions please don't hesitate to contact the creator of this issue.

## Checklist/directions

- [ ] Mantid opens without errors or warnings 
- [ ] Every option in `Interface` opens a GUI 
- [ ] Load some test data 
- [ ] Catalogue opens  [directions here](https://www.mantidproject.org/Catalog_Search) 
- [ ] Saving/loading projects works  [directions here](https://www.mantidproject.org/MantidPlot:_The_File_Menu#File-.3E_Save_Project_as...)

NickDraper, 2019-07-19:
Tested and passed on windows 7 mantidplot

NickDraper, 2019-07-19:
Tested and passed on IDAaaS mantidplot and workbench

Harrietbrown, 2019-07-19:
Tested and passed on windows 7 mantidplot and workbench

ConorMFinn, 2019-07-19:
Tested and passed on Ubuntu 18.04 || MantidPlot and Workbench

alicerussell1, 2019-07-22:
Tested on Windows 10 mantidplot and workbench - no unknown issues found

martyngigg, 2019-07-23:
Tested on OSX 10.12 after latest changes for embedded help - all looks good.

NickDraper, 2019-07-23:
tested on windows 7 mantidplot and workbench

martyngigg, 2019-07-23:
Tested on RHEL 7 and all looks fine.

DTasev, 2019-07-23:
Testing on Ubuntu 18.04

closed 2019-07-23 by NickDraper