mantidproject/mantid #26412: Plotting Smoke Tests

NickDraper, 2019-07-19:
# Plotting smoke tests

Please run these tests on the compiled package of Mantid; **not a locally compiled version**.

When reporting back please include:

- Your OS
- Your facility/location
- If you have other package versions of Mantid installed (yes/no)

If you have any questions please don't hesitate to contact the creator of this issue.

## Checklist/directions

* 1D plotting:[instructions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/03_displaying_1D_data.html#displaying-1d-data)
	- [ ] Simple plot 
	- [ ] Another way to plot 
	- [ ] Adding curves to existing plots 
* 2D plotting: [instructions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/04_displaying_2D_data.html#displaying-2d-data)
	- [ ] Plot all spectra 
	- [ ] Plot from matrix 
	- [ ] Change colour map 
	- [ ] Contour plot 
	- [ ] Spectrum viewer 
* 3D plotting:
	- Load some data eg `LOQ7041` from the ISIS sample data
	- Double click the workspace to open the table view
	- Select the table
	- [ ] 3D plot becomes available as an option in the toolbar 
	- [ ] 3D wire frame 
	- [ ] 3D hidden line 
	- [ ] 3D polygon 
	- [ ] 3D wire surface

ConorMFinn, 2019-07-19:
**All Tests Passed.**
**OS:** Ubuntu 18.04
**Facility:** ISIS
**Other Versions:** Yes

alicerussell1, 2019-07-19:
Tested on windows 10, no issues found

NickDraper, 2019-07-19:
tested on iDAaaS, workbench passed

closed 2019-07-22 by NickDraper